chris1111/Wireless-USB-Adapter-Clover #1: Please add support Ralink/Mediatek RT series WiFi USB

UniqueC0de, 2018-03-27:
Ralink/Mediatek RT2870/ RT2770/ RT3X7X/ RT537X/ RT5572

chris1111, 2018-03-27:
This Drivers its here https://github.com/chris1111/TP-LINK-WIFI-Network-macOS

UniqueC0de, 2018-03-27:
so you mean I can use driver from [https://github.com/chris1111/TP-LINK-WIFI-Network-macOS](url), and use [https://github.com/chris1111/Wireless-USB-Adapter-Clover](url) configuration utility?

chris1111, 2018-03-27:
No the utilities are diferant

UniqueC0de, 2018-03-27:
Ok, I want is after configure the internet connection.
I want the internet to be connected without configuration utility to be visible when macOS startup, or when we tap on Turn WiFi On button.

This configuration utility is kind of annoying that appears each time when I start macOS.

Anyway, thanks for your reply.

chris1111, 2018-03-27:
Yes this utility is different, utility to be visible when macOS startup but its not a big deal

closed 2018-03-27 by chris1111