chris1111/Wireless-USB-Adapter-Clover #3: Installing on a "real" Mac

GJNilsen, 2018-11-06:
I'm having problems with a USB wifi dongle after installing Mojave on an Xserve and would like to try these drivers. How does one install it on a real Mac?

chris1111, 2018-11-08:
Ok please test this package for real Mac
I test on my Hackintosh MacBook Pro and its work properly
[image]

[image]

[image]

[Wireless USB Adapter.zip](https://github.com/chris1111/Wireless-USB-Adapter-Clover/files/2562088/Wireless.USB.Adapter.zip)

remember I am not responsable anny boot failled on real Mac or Hackintosh For this software
You use it at your own risk

adzodamelk, 2018-11-08:
I've just tested and yessssssss, it runs smoothly on my MacBook Air 2012 with Mojave 10.14.1!!!! 

Thank you so much again Chris!

chris1111, 2018-11-08:
> I've just tested and yessssssss, it runs smoothly on my MacBook Air 2012 with Mojave 10.14.1!!!!
> 
> Thank you so much again Chris!

You are tes the package available here?
please confirm and I will make available in release

adzodamelk, 2018-11-08:
> You are tesing the package available here?
> please confirm and I will make available in release

Today morning I've tested successfully the second one you published yesterday, I mean
https://github.com/chris1111/TP-LINK-WIFI-Network-macOS/issues/4#issuecomment-436652761

The first one you sahred with me (https://github.com/chris1111/TP-LINK-WIFI-Network-macOS/issues/4#issuecomment-436425200) didn't run, at least with my dongle TP-LINK TL-WN823N v2, the error message was "no wifi usb connected"

chris1111, 2018-11-08:
> > You are tesing the package available here?
> > please confirm and I will make available in release
> 
> Today morning I've tested successfully the second one you published yesterday, I mean
> [chris1111/TP-LINK-WIFI-Network-macOS#4 (comment)](https://github.com/chris1111/TP-LINK-WIFI-Network-macOS/issues/4#issuecomment-436652761)
> 
> The first one you shared with me ([chris1111/TP-LINK-WIFI-Network-macOS#4 (comment)](https://github.com/chris1111/TP-LINK-WIFI-Network-macOS/issues/4#issuecomment-436425200)) instead didn't run, at least with my dongle TP-LINK TL-WN823N v2; the error message was "no wifi usb connected".

Please test the one here its more clean, Uninstalled the one you have installed reboot and test this one
thanks

adzodamelk, 2018-11-08:
> Please test the one here its more clean, Uninstalled the one you have installed reboot and test this one
> thanks

Just done as you asked! Also this one runs like a charme, the only strange thing is that it took much more time to install than the second one you shared yesterday.

Thanks again!!

chris1111, 2018-11-08:
> > Please test the one here its more clean, Uninstalled the one you have installed reboot and test this one
> > thanks
> 
> Just done as you asked! Also this one runs like a charme, the only strange thing is that it took much more time to install than the second one you shared yesterday.
> 
> Thanks again!!

Thank you very mutch 
yes it take more time because is install directly in S/L/E with the correct permission and rebuild kernel caches. I will make a release for real Mac and editing the page Hope the @GJNilsen will try it to

adzodamelk, 2018-11-08:
Thank to you for so much for sharing your work! 
Without it, I would have to roll back from Mojave to Sierra in order to use TP-Link official driver....

GJNilsen, 2018-11-08:
Hi @chris1111 , thanks a lot for the driver. I did not have time to wait for it, so I botched up my own, but Ill try yours later on. Thanks again! 👍

chris1111, 2018-11-08:
When I get back home tonight I Will test on Mavericks and If its work like it should I will Create new page exclusively for this program

chris1111, 2018-11-09:
Ok guys the new page here https://github.com/chris1111/Wireless-USB-Adapter

urbanman2004, 2018-11-09:
@chris1111 how's it going man, long time. It's frank Nitty from InsanelyMac

chris1111, 2018-11-09:
> @chris1111 how's it going man, long time. It's frank Nitty from InsanelyMac. Had been following your topic an received notification

Great! thanks frank  
you are you doing good