chris1111/Wireless-USB-Adapter-Clover #5: Can you please add support for TP-Link WN821N V4?

Lolihunter1337, 2018-12-25:


chris1111, 2018-12-25:
> No description provided.

Used latest release and change Drivers on Clover
let me known if that working
[Archiver.zip](https://github.com/chris1111/Wireless-USB-Adapter-Clover/files/2707890/Archiver.zip)

Lolihunter1337, 2018-12-25:
is it possible to try on a VM ware first? so using the "real mac" version?

chris1111, 2018-12-25:
> is it possible to try on a VM ware first? so using the "real mac" version?

try
 I don't know, I am not use VM

Lolihunter1337, 2018-12-25:
doesnt work. it gets detected but doesnt show any Network i also tried https://github.com/chris1111/Wireless-USB-Adapter/releases/tag/V2 but same problem

chris1111, 2018-12-25:
> doesnt work. it gets detected but doesnt show any Network i also tried https://github.com/chris1111/Wireless-USB-Adapter/releases/tag/V2 but same problem

Thanks to report I think support WN821N V4 is drop after Yosemite 10.10

Lolihunter1337, 2018-12-25:
Got it working on Sierra 10.12 thanks man

closed 2018-12-25 by chris1111