chris1111/Wireless-USB-Adapter-Clover #7: Issue with 10.14.3

profzei, 2019-02-06:
I've bought a TL-WN823Nv3 wireless N usb adpater.
It was working like a charm with hackintosh 10.14. I updated my machine following this guide https://www.reddit.com/r/MatebookXPro/comments/am046y/its_february_bois_hackintosh_feb_update/?sort=new
... and then I uninstalled the package and then re-installed the package Wireless USB Adapter Clover-V4.pkg but the green led on the usb wireless adapter remains off: it don't work...
Any help will be very appreciated
Thank you

chris1111, 2019-02-06:
What is the Clover revision because somes Clover revision before 4871 inject kext is broken 
Also try to un install and reinstall the Utility 
No issue here with 3 adapter on Mojave 10.14.3

chris1111, 2019-02-24:
See here problem is fix
[
](https://github.com/chris1111/Wireless-USB-Adapter-Clover/issues/9)

closed 2019-02-24 by chris1111