chris1111/Wireless-USB-Adapter-Clover #9: Kernel panic on mojave 10.14.3

maz-1, 2019-02-23:
My WLAN device is COMFAST CF-916AC (RTL8811AU)
As soon as I try to connect to a 2.4ghz hotspot the system panics
Panic log: https://gist.github.com/maz-1/b155930f15948089519902dcf2d3d443

chris1111, 2019-02-23:
> My WLAN device is COMFAST CF-916AC (RTL8811AU)
> As soon as I try to connect to a 2.4ghz hotspot the system panics
> Panic log: https://gist.github.com/maz-1/b155930f15948089519902dcf2d3d443

latest Release of Mojave update is 10.14.3 (18D109) no issue here

chris1111, 2019-02-23:
OK try Release V5 but first of all, uninstall the V4
https://github.com/chris1111/Wireless-USB-Adapter-Clover/releases/tag/V5

maz-1, 2019-02-24:
> OK try Release V5 but first of all, uninstall the V4
> https://github.com/chris1111/Wireless-USB-Adapter-Clover/releases/tag/V5

Great, it works.

closed 2019-02-24 by maz-1