chris1111/Wireless-USB-Adapter-Clover #11: WiFi adapter will not reconnect after restart

ZhuJo, 2019-03-06:
After restart/shutdown of hackintosh with macOs 10.14.3 (v5 of drivers) adapter will not automatically connect to hidden WiFi network (WPA2 AES) it was previously connected to.

chris1111, 2019-03-06:
> After restart/shutdown of hackintosh with macOs 10.14.3 (v5 of drivers) adapter will not automatically connect to hidden WiFi network (WPA2 AES) it was previously connected to.

Thanks
what should you do to connect it ?
otherwise use the v4 if it works best for you

ZhuJo, 2019-03-06:
behaviour was the same with v4.

I have to again enter network name and password. Connecting to Time Capsule.

Jozef

chris1111, 2019-03-06:
which adapter you use

ZhuJo, 2019-03-06:
it's noname adapter based on Realtek 8812AU.
From System Profiler:
USB3.0 802.11ac 1200M Adapter:

  Product ID:	0xb812
  Vendor ID:	0x0bda  (Realtek Semiconductor Corp.)
  Version:	2.10
  Serial Number:	123456
  Speed:	Up to 480Mb/sec
  Manufacturer:	Realtek
  Location ID:	0x14400000 / 11
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

ZhuJo, 2019-03-07:
You can close the issue, I have solved it by installing Mac drivers instead. 
Clover drivers are behaving bad since beginning – first of all, your installer can't mount EFI partition and thus install drivers to /EFI/CLOVER/kexts/others so it has to be done manually. 

Also, folder WLAN in /Library/Application Support has wrong permissions and is readable only by root.

closed 2019-03-07 by chris1111