chris1111/Wireless-USB-Adapter-Clover #12: Can you please add support for Realtek 8821CU

weachy, 2019-03-12:
Hi,
I have a "Realtek 8821CU Wireless USB Adapter". Can you please add support it? 
Here is a driver that works fine: [https://www.ezcast.com/res/app/EZC-5200-MacOS10.9_Ma.zip](url).
Maybe it's newer than yours? I wish it would be helpful. 
Thank you for your hard working.

chris1111, 2019-03-12:
> Hi,
> I have a "Realtek 8821CU Wireless USB Adapter". Can you please add support it?
> Here is a driver that works fine: [EZC-5200-MacOS10.9_Ma.zip](https://www.ezcast.com/res/app/EZC-5200-MacOS10.9_Ma.zip).
> Maybe it's newer than yours? I wish it would be helpful.
> Thank you for your hard working.

I test and its work here

Please test but before uninstall your drivers with your utility then reboot and Install with my program 

[Wireless USB Adapter Clover-V6.zip](https://github.com/chris1111/Wireless-USB-Adapter-Clover/files/2956559/Wireless.USB.Adapter.Clover-V6.zip)

weachy, 2019-03-13:
V6 is working like a charm. thx! 👏

chris1111, 2019-03-13:
> V6 is working like a charm. thx! 👏

Thanks I will release V6 👍🏻

closed 2019-03-13 by chris1111