chris1111/Wireless-USB-Adapter-Clover #13: Constant WPS scan

ZhuJo, 2019-03-12:
When I'm connected to my (non WPS) network, quite often a windows scanning for WPC compatible AP appears and connection is interrupted until I turn off and on WiFi.

Is it possible to completely disable WPS scanning? I'm not interested at all to use it, nor I do have AP that allows me to enter WPS PIN.

chris1111, 2019-03-13:
Take a look here maybe that help?
https://www.lifewire.com/disable-wps-on-router-4137379

ZhuJo, 2019-03-13:
That's not a solution, Realtek drivers are showing that on macOs by itself. I DON'T have any WPS capable router, nor do I ever want to use that feature so I don't know why it's showing at all – again, this is by your drivers itself, has nothing to do with anything else.
Only thing I'm asking about is to be able to turn off that forever in your connection utility.

chris1111, 2019-03-13:
> That's not a solution, Realtek drivers are showing that on macOs by itself. I DON'T have any WPS capable router, nor do I ever want to use that feature so I don't know why it's showing at all – [...]

Its not my Driver its a Realtek Driver 
Sorry but I think your issue is cause by your Rooter, I never hear things like that
issue is closed

closed 2019-03-13 by chris1111