chris1111/Wireless-USB-Adapter-Clover #19: AC56 Issues

daniel-farina, 2019-06-21:
Hello Guys, 

I'm trying to make this network adapter work.  
https://www.amazon.com/gp/product/B00FB45USW/
I see "SUS_USB-AC56_8812AU" on the list of supported devices but I installed it and no luck. 
What does the AU stand for? Australia, I live in US would the model change?  Not sure where to find the full model number. 

I also tried with this one and no luck.
https://www.amazon.com/dp/B0159222VS/ 

I'm running Mojave 10.14.5 with the latest Clover boot loader.  I do see the RtWlanU and RtWlanU1827 in my kexts/other folder inside my EFI bootloader. 

I did see the network connections for half a second at one point and then it never worked again. I do no see the app asking for permission after the install. 

Any help would be appreciated

Thank you

chris1111, 2019-06-21:
Sorry but your not the only one with an issue with the ASUS adapter and I really don't know why??? 
The PCI Expres card Rosewill RNX-AC1900PCE is not support

Maybe try a different release for your Asus adapter, let me know if one of them work

closed 2019-06-21 by chris1111

daniel-farina, 2019-06-21:
Update:
I got it work with the drivers listed on the official page. I had installed the clover drivers and then installed the utility from here for MacOs: https://www.asus.com/us/Networking/USBAC56/HelpDesk_Download/

It looks like it overwrote the BarApp and maybe a few other things:
[image]

I also added the app to the app to the accessibility settings, not sure if that helped:
[image]

chris1111, 2019-06-21:
But you have a double Drivers, your ASUS Utility installed drivers in S/L/E and I am pretty sure its work with the lattes Drivers here from Clover Your Utility works for El Capitan