chris1111/Wireless-USB-Adapter-Clover #20: Can't see the network list, but can connect network by add SSID and password

ourfor, 2019-06-29:
- The version of current operation system is ` macOS catalina 10.15 beta `, the version of ` Wireless-USB-Adapter-Clover ` is ` Wireless USB Adapter Clover-V7 ` 

- I follow the steps on github readme ` Instructions `

> here is the screenshot 

[image]

[image]

[image]

chris1111, 2019-06-29:
Thats really strange ! Looks like a bug in Catalina beta ??

ourfor, 2019-06-29:
I don't know。😅

closed 2019-07-06 by chris1111

daniel-farina, 2019-07-10:
Same here with Mojave: 

The list of networks does not show but manually entering the network name and password works. 

for the AC56 I used the network name and then WPA_2 for authentification type.

Thank you

blackadmin, 2019-10-08:
On Mojave facing same issue, connecting fine but no scan results. :(