chris1111/Wireless-USB-Adapter-Clover #32: V7 - nano t2u

madhav0922, 2019-09-15:
Whenever i plugin my tp-link t2u nano adapter, the system just runs into a kernel panic and reboots straight away.
i am able to reach the password pane though. As soon as i enter my password and hit enter, it reboots

chris1111, 2019-09-15:
On what macOS? I never heard a problem like this
SIP is désactivate ?

madhav0922, 2019-09-15:
I am using a hackintosh.
Mac os Mojave 10.14
On v6 i had the tp link TL-WN725N which works great on both v6 and v7 unline t2u nano

chris1111, 2019-09-15:
> I am using a hackintosh.
> Mac os Mojave 10.14.3
> On v6 i had the tp link TL-WN725N which works great on both v6 and v7 unlike t2u nano

Why not Updating macOS to to 10.14.6 ?

madhav0922, 2019-09-15:
Because for some reason it gets stuck while updating. I'll have to redo mac from scratch to update..so when i'll get time i'll update.
Will updating fix the issue though? So i'll report after updating.
One more thing that with the default driver provided by tp-link on their official site, t2u nano is working.

closed 2019-09-15 by chris1111