chris1111/Wireless-USB-Adapter-Clover #34: TL-WN725N v2 compatibility?

kraftpunk97, 2019-10-10:
Hi. I have a TL-WN725Nv2 adapter. Has this model been tested in any capacity?

chris1111, 2019-10-10:
> Hi. I have a TL-WN725Nv2 adapter. Has this model been tested in any capacity?

yes is been tes and its work from 10.9 to 10.15

closed 2019-10-10 by chris1111

x43x61x69, 2019-10-14:
Just FYI that TL-WN725N v2 works with v8 on Catalina final version (19A583).

However, I can't get the available networks to show. Manually enter the SSID and password is the only way.

kraftpunk97, 2019-10-14:
@chris1111 This seems like a minor bug. Can this be fixed?

lj19940623, 2019-12-08:
TL-WN725N v2 works with v11 on Catalina. But bug mentioned by x43x61x69 still here. Please fix...@chris1111 @kraftpunk97

chris1111, 2019-12-08:
Uninstall the Wireless USB Adapter Clover version and try this [Wireless USB Adapter](https://github.com/chris1111/Wireless-USB-Adapter)