chris1111/Wireless-USB-Adapter-Clover #35: Support for EDUP RTL8821CU-CG chipset  (Wifi + BT)

madhankumar388, 2019-10-11:
Hi Chris GM .. Just checking to see if you are planning to add the support for RTL8821CU-CG chipset anytime in the near future ? . I am trying to use it on my hackintosh 10.15 Catalina . I purchased it on ebay using the below link . This dongle also has BT built in, anyidea if BT & Wifi will work after support for this chip-set is added ?. 

Thank you and have a Nice day

Below are the tech specs of the item from EDUP manufacturers website :
http://www.szedup.com/product-item/2-1-bluetooth-wireless-adapter-600mbps-wifi-direct-nano-usb-adapter/

Below are the link to its drivers from the manufacturers website :
http://www.szedup.com/ep-ac1661/

Ebay Purchase link :
https://www.ebay.com/itm/USB-WiFi-Bluetooth-Adapter-Dual-Band-ac600Mbps-EDUP-USB-2-0-2-4G-5-8G/264479279193?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

chris1111, 2019-10-11:
Did you test my program with your Dongle?
maybe works?

closed 2019-10-11 by chris1111

madhankumar388, 2019-10-11:
I installed the  Wireless USB Adapter Clover-V8.pkg on my catalina hackintosh . i used the vanilla guide and it is set to enable kexts on boot . but the dongle is not getting detected . when i plug it into windows PC the led on dongle lights on . when i plug into my hackintosh nothing happens . no led lights also . tried both usb 3.0 and 2,0 ports , same result .