chris1111/Wireless-USB-Adapter-Clover #39: Nexxt Lynx 600-ac not supported

alex-spataru, 2019-10-28:
Hello,

I just found this project because the driver for the [Lynx 600-ac USB adapter](https://www.nexxtsolutions.com/connectivity/us_en/aulub605u1) is not supported under Catalina (I think this has to do with 32-bit support being dropped). However, when installing this project and configuring Clover's config.plist, the USB device is not recognized by this project. 

Although I have zero experience developing Kext's, I do have some experience with C/C++ and am willing to help support this device (if possible). 

Output of `ioreg -p IOUSB`: `802.11n WLAN Adapter@14200000  <class AppleUSBDevice, id 0x10000056a, registered, matched, active, busy 0 (1 ms), retain 11>`

Can something be done to support this device, or do you have any tips?

Thanks!

chris1111, 2019-10-28:
Sorry but is not support; your Device use a 32 bit app and its not on the list of support

closed 2019-10-28 by chris1111

alex-spataru, 2019-10-28:
Thanks for the feedback, I've just sent an email to the provider. If they respond to me, or provide a supported driver, I'll let you know through here.