chris1111/Wireless-USB-Adapter-Clover #41: RTL 8187 Not working with Wireless Utility.

Vikesyy, 2019-11-08:
Hello,

macOS Mojave recognized my wifi, but your Utility doesn't show my device nor the setting. Please I request you to add functionality for my device RTL 8187. I will be grateful.

[image]

Thank You

chris1111, 2019-11-08:
of course not working your adapter not using the same drivers and its not support by the utility
You are using rtl8187.kext
[image]

closed 2019-11-08 by chris1111

Vikesyy, 2019-11-08:
I didn’t use any Kext, it recognised it. And how my device is RTL 8187, why isn’t it working?

chris1111, 2019-11-08:
it is recognized if you install the package that corresponds to your usb adapter.