chris1111/Wireless-USB-Adapter-Clover #42: TL-WN725N 3.0 免驱版显示 No USB-WiFi Plug in

lable, 2019-11-10:
TL-WN725N 3.0 免驱版显示 No USB-WiFi Plug in

chris1111, 2019-11-10:
Change USB port then reboot
Also veriying config.plist
,### View ➣ [➤ You must have set injectkexts to Yes in the Clover config.plist](https://i25.servimg.com/u/f25/18/50/18/69/captu819.png)

closed 2019-11-10 by chris1111

lable, 2019-11-10:
config.plist -> 

	<dict>
		<key>CustomUUID</key>
		<string>FB78CF39-348B-441A-91A9-44157C41B953</string>
		<key>InjectKexts</key>
		<string>Yes</string>
		<key>InjectSystemID</key>
		<true/>
	</dict>

nxnyk, 2020-01-12:
> config.plist ->
> 
> ```
> <dict>
> 	<key>CustomUUID</key>
> 	<string>FB78CF39-348B-441A-91A9-44157C41B953</string>
> 	<key>InjectKexts</key>
> 	<string>Yes</string>
> 	<key>InjectSystemID</key>
> 	<true/>
> </dict>
> ```

老哥你解决了吗？