chris1111/Wireless-USB-Adapter-Clover #46: Request support with RTL8811CU

shm007g, 2019-12-02:
RTL8811CU is similar with RTL8811AU as I see. But RTL8811CU not in the support list. Could you please check and support RTL8811CU as well?

Much appreciated !

chris1111, 2019-12-02:
> RTL8811CU is similar with RTL8811AU as I see. But RTL8811CU not in the support list. Could you please check and support RTL8811CU as well?
> 
> Much appreciated !

This Driver is made foe Windows
See official Realteak drivers
[For Realtek USB WLAN Family Controller.](https://www.realtek.com/en/component/zoo/category/rtl8811cu-software)

closed 2019-12-02 by chris1111

shm007g, 2019-12-02:
I mean I could use this RTL8811CU(CF-811AC) on macOS 10.14 with official driver which may fail for its 32-bit-build. Could you make this RTL8811CU work on Catalina?

[image]

chris1111, 2019-12-02:
please upload driver package here

shm007g, 2019-12-02:
[RTLWlanU_MacOS10.9_MacOS10.14_Driver_1830.20.b31_1827.4.b32_UI_5.0.8.b4.zip](https://github.com/chris1111/Wireless-USB-Adapter-Clover/files/3910782/RTLWlanU_MacOS10.9_MacOS10.14_Driver_1830.20.b31_1827.4.b32_UI_5.0.8.b4.zip)

Really Appreciated !

chris1111, 2019-12-02:
[Did you try latest release  ](https://github.com/chris1111/Wireless-USB-Adapter-Clover/releases/tag/V11)

shm007g, 2019-12-10:
@chris1111  I tried last weekend, it works fine. Really thx.