chris1111/Wireless-USB-Adapter-Clover #47: Can't verify Mac App Store apps on Catalina

aerz, 2019-12-06:
I've already upgraded from Mojave to Catalina and installed the latest version of the drivers to make work my TP-Link TL-WN725N dongle. After that, I don't be able to open the apps purchased on the App Store which throws me an error all the time.

[image]

I made a short research and I found some comments about the network interface must be en0 to allow check the Apple certificate. I've tried a lot of things to change en1 (default interface installed by the installer script) to en0 but I could not make it works.

[image]

- [An iDiot's guide to iMessage](https://www.tonymacx86.com/threads/an-idiots-guide-to-imessage.196827/)

> You must have a working Ethernet connection which is defined as BuiltIn and en0.

-  [Certificate problems causing app authentication errors for some Mac App Store users](https://appleinsider.com/articles/15/11/12/certificate-problems-causing-app-authentication-errors-for-some-mac-app-store-users) - This is the certificate that MAS is not able to verify.

I posted this issue here because I thought there is a way to modify the network interfaces but I don't know how to do it.

chris1111, 2019-12-06:
This Utility as nothing to do with your issue
sorry but issue is closed

closed 2019-12-06 by chris1111

coderobe, 2019-12-22:
for posterity: if you do not have a builtin ethernet chipset you need [NullEthernet](https://github.com/RehabMan/OS-X-Null-Ethernet) for this to work.