chris1111/Wireless-USB-Adapter-Clover #48: TL-WN823N not recognized

NotDaba, 2019-12-17:
Hello.
First of all, thanks for the work you are doing.
My USB WIFI can't be recognised.
Maybe it's because it's V3 but not sure about this. 
[image]

chris1111, 2019-12-17:
Follow the guide correctly maybe a reboot is neneccesary but the WN823N V3 work and is already test by lots of people

closed 2019-12-17 by chris1111

NotDaba, 2019-12-18:
I'm really sure about having followed all the steps at 100%.
Tried all my USB port on notebook, tried also lot of restart.
[image]

Mr-Bubble, 2019-12-22:
Hot boot from windows to Max, it works，You try

NotDaba, 2019-12-22:
This worked. 
Thanks a lot. 
--
Inviato da Libero Mail per Android domenica,

Mr-Bubble, 2019-12-22:
You're welcome, cold boot doesn't work, this is obviously a bug, hope the author can fix it

NotDaba, 2019-12-22:
Yea! Thanks a lot for real! 
--
Inviato da Libero Mail per Android domenica,

Mr-Bubble, 2019-12-22:
@chris1111