chris1111/Wireless-USB-Adapter-Clover #51: During booting and shutdown,  prints "<=== HalUsbInMpdu()" often.

dnhkng, 2019-12-23:
Hi,

After installing a T2U Nano dongle, I see the message:
`<=== HalUsbInMpdu()`

printing about 10 times a second during bootup and shutdown. Is this normal behavior?

chris1111, 2019-12-23:
Yes this message is normal in verbose mode -V

closed 2019-12-23 by chris1111

lazd, 2020-01-05:
This actually prints non-stop, even after boot, several times per second. This message should be disabled.

chris1111, 2020-01-05:
The release V11 its a first official beta driver for Catalina so If your not happy with the release V11 use release V10 this message not exist
https://github.com/chris1111/Wireless-USB-Adapter-Clover/releases/tag/V10

chris1111, 2020-02-26:
Issue is resolved with the [Release V13](https://github.com/chris1111/Wireless-USB-Adapter-Clover/releases/tag/v13)