chris1111/Wireless-USB-Adapter-Clover #53: TL-WN823N not recognized

Mr-Bubble, 2019-12-24:
cold boot doesn't work, but Hot boot from windows to Mac, it works，plese fix it

chris1111, 2019-12-24:
https://github.com/chris1111/Wireless-USB-Adapter-Clover/issues/48

closed 2019-12-24 by chris1111

Mr-Bubble, 2019-12-25:
@chris1111 cold boot doesn't work, this is obviously a bug, hope the author can fix it

chris1111, 2019-12-25:
> @chris1111 cold boot doesn't work, this is obviously a bug, hope the author can fix it

what do you mean by cold boot ?

Mr-Bubble, 2019-12-25:
Boot from shutdown status to Mac it no work，boot from windows to Mac, it works

chris1111, 2019-12-25:
never heard of that, **I have 5 adapters** and it all works whether I reboot or do a Shotdown and I also have a dual boot windows 10

Mr-Bubble, 2019-12-25:
Not only I have this problem, #48 also have this issue

chris1111, 2019-12-25:
> Not only I have this problem, #48 also have this issue

I have the TL-WN823N I will show you some things today

chris1111, 2019-12-25:
## Video ⬇︎
[image]

So this the prouve of working check issue on your System