chris1111/Wireless-USB-Adapter-Clover #62: OpenCore Bootloader

NotDaba, 2020-01-06:
Can we have an installer for OpenCore?

chris1111, 2020-01-06:
no

closed 2020-01-06 by chris1111

dorald, 2020-01-20:
I just managed to install (manually) on OpenCore. Just extract the two Kexts from this clover installer and copy to /EFI/OC/Kexts then edit config.plist to add the two kexts under Kernel -> Add
Worked for me :)

vimcaw, 2020-03-22:
> I just managed to install (manually) on OpenCore. Just extract the two Kexts from this clover installer and copy to /EFI/OC/Kexts then edit config.plist to add the two kexts under Kernel -> Add
> Worked for me :)

@Dorald Thank you for sharing, but I only get some strange files, how do you extract the kexts?

The content of clover installer: 

[image]

The content of each *.pkg file: 

[image]

The Resources folder only has some text files.

chris1111, 2020-03-22:
see video here
https://github.com/chris1111/macOS-Extractor

vimcaw, 2020-03-22:
@chris1111 Got it 😀, Thanks!

tanpengsccd, 2020-12-22:
after install pkg, U find ESP /EFI/CLOVER/kexts/Other RtWlanU1827.kext, RtWlanU.kext , and manual config OC config.plist