chris1111/Wireless-USB-Adapter-Clover #64: After sleep wifi so slowly

dobpilot, 2020-01-07:
1. My ASUS AC-68 connect by internal USB 3.0 port ( Z170 MSI MATE board )
2. USB 3.0 in hackintosh, works by patched SSDT and codeless general usb3 kext. I use USBMap script, https://github.com/corpnewt/USBMap
3. Wifi works perfectly avr download and upload speed is 300 - 400 megabits per second
4. But after sleep download speed does not exceed 50 megabits and upload speed about 200 megabits

chris1111, 2020-01-07:
There is nothing to do regarding USB 3 on Hackintosh; not less to my knowledge

closed 2020-01-07 by chris1111

dobpilot, 2020-01-07:
Maybe it is possible to increase the debug information of the driver or show the driver sources?

chris1111, 2020-01-07:
> Maybe it is possible to increase the debug information of the driver or show the driver sources?
the source is not available, its not open source but when Asus release the driver for Catalain you can post about the issue on Asus Forum ?
https://rog.asus.com/forum/