chris1111/Wireless-USB-Adapter-Clover #69: Issues while requires username and passwd

billhhh, 2020-02-20:
My uni has public wifi which requires username and passwd, however the adapter does not working...

chris1111, 2020-02-20:
Elaborate please I don't understand your issue

closed 2020-02-20 by chris1111

billhhh, 2020-02-20:
Hi Chris,

Thank you for your rapidly reply. Sorry for not elaborating the question clearly.

The adapter works in most of my situations. But the public wifi hotspot of my uni requires username and passwd, using a different Encryption type --- EAP. After I typed in the username and passwd, it disappeared.
BTW, I am using a relatively old version, May I know the lastest version has fixed it or there is a way to access the EAP encrypted network?

Thank you so much!

chris1111, 2020-02-21:
See here
https://github.com/chris1111/Wireless-USB-Adapter-Clover/issues/37

billhhh, 2020-02-21:
Thank you, Chris