chris1111/Wireless-USB-Adapter-Clover #70: drivers installed but No USB-WIFI Plug in

seekhust, 2020-03-22:
Hi,
I have installed the wireless USB adapter V11 for catalina 10.15.2.

I found it's interesting.

I can see the hardware of the USB wifi under the system report-hardware-USB device tree.

But The USB WIFI adapter can't be recognized on the tool bar. When I click the wifi icon, it say "No USB-WIFI Plug in"

How it happened?

THanks.

chris1111, 2020-03-22:
which adapter?

seekhust, 2020-03-22:
The USB WIFI adapter.

You can see it in http://www.comfast.cn/index.php?m=content&c=index&a=show&catid=13&id=74

chris1111, 2020-03-22:
### View ➣ [➤ You must have set injectkexts to Yes in the Clover config.plist](https://i25.servimg.com/u/f25/18/50/18/69/captu819.png)

closed 2020-03-23 by chris1111

seekhust, 2020-03-23:
Thanks, but I have chosen inject Kexts to Yes, and tick the inject system ID.
The error remains the same.

"No USB-WIFI plug in"

chris1111, 2020-03-23:
> Thanks, but I have chosen inject Kexts to Yes, and tick the inject system ID.
> The error remains the same.
> 
> "No USB-WIFI plug in"

So Comfast adaptor is not support sorry

seekhust, 2020-03-24:
After I installed the driver, I can't see two kext files :RtWlanU827.kext and RtWlanU.kext under "ESP/EFI/CLOVER/kexts/other" folder.

So I copied these two files,
It works now.

Thanks.