chris1111/Wireless-USB-Adapter-Clover #71: works fine-but only if booted with dongle

Pedroooow, 2020-03-23:
Hey ! 
Merci Chris pour ce programme bien pratique ;) 

Just 1 question: it works fine with my wifi dongle but only if I boot with the dongle plugged in. 

If I boot connected with ethernet at my desktop and decide to unplug my laptop to go work elsewhere, I have to reboot the computer after having plugged the dongle. 

Is there any way to refresh the software / kill it and relaunch without having to reboot ? 

Thanks !

chris1111, 2020-03-23:
Thanks
I don't know any way to get there, if you find come tell us .
Thank you for bringing it.

hacker1024, 2020-04-06:
This is odd, it seems to work fine when the kexts are installed the normal way (not through Clover).
Can we leave this issue open? It hasn't been resolved yet.

chris1111, 2020-04-06:
> This is odd, it seems to work fine when the kexts are installed the normal way (not through Clover).
> Can we leave this issue open? It hasn't been resolved yet.
yes it is true that it works with this one [Wireless USB Adapter](https://github.com/chris1111/Wireless-USB-Adapter)  but what does it change whether this issue is closed or open

chris1111, 2020-06-02:
Fix
https://github.com/chris1111/Wireless-USB-Adapter-Clover/releases/tag/V15

closed 2020-06-02 by chris1111