chris1111/Wireless-USB-Adapter-Clover #74: macOS laggs when connecting dongle

marcinbauer85, 2020-04-11:
Hi,
I have a TP link wn722N (rev 3.0), and when connecting it into my Hackintosh 10.15.4 everything starts to lag. I've tried the official TP drivers, and yours clover/normal, but it happens all the time.
Do you have any idea why this might be?
This is my sytem report, and for hardware I have:
GA-z77-d3h, i7-3770k (stock), 16GB ram, R5700XT
System report: https://we.tl/t-2ktDphda5X
Clover config: https://we.tl/t-X0mVgTVQqy

As a note, my Hackintosh is working perfectly, but this dongle makes it pretty much unusable.
Thanks for the help!

chris1111, 2020-04-11:
Use only one Package !
 remove official TP drivers or remove Wireless USB adapter Clover

closed 2020-04-11 by chris1111

marcinbauer85, 2020-04-11:
Yes I did uninstall the official, and yes I used just one package at a time.

chris1111, 2020-04-11:
I don't think its relate to the WIFI utility, check your Clover Folder ? I never see macOS freeze because of the Realtek USB wifi.

chromer030, 2020-04-11:
Install kexts in L/E , because the way that Clover inject kexts break system integrity and deeply impact on system stability.

marcinbauer85, 2020-04-12:
I tried also to remove my AtherosE2200Ethernet.kext since I was using an ethernet connection before with the same results. My Hackintosh is very laggy, and its unsuable. I also tried different USB ports, and the same problem.
I have one difference in the setup as opposed to yours. When Im in the network settings, I don't see a Wifi option, just Ethernet https://ibb.co/3RTttB0
Plus notice that the 'kernel_task' i taking up most of my CPU.

Is there any diagnostic that I can send you to help me with this?

marcinbauer85, 2020-04-12:
This is my NetworkInterfaces.plist after I reset it:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Interfaces</key>
	<array>
		<dict>
			<key>Active</key>
			<true/>
			<key>BSD Name</key>
			<string>en0</string>
			<key>IOBuiltin</key>
			<false/>
			<key>IOInterfaceNamePrefix</key>
			<string>en</string>
			<key>IOInterfaceType</key>
			<integer>6</integer>
			<key>IOInterfaceUnit</key>
			<integer>0</integer>
			<key>IOMACAddress</key>
			<data>
			UD6qio/h
			</data>
			<key>IOPathMatch</key>
			<string>IOService:/IOResources/AppleUSBHostResources/AppleUSBLegacyRoot/AppleUSBXHCI@14000000/802.11n NIC@14300000/AppleUSBInterface@0/RtWlanU/en0</string>
			<key>SCNetworkInterfaceInfo</key>
			<dict>
				<key>UserDefinedName</key>
				<string>Ethernet Adaptor (en0)</string>
			</dict>
			<key>SCNetworkInterfaceType</key>
			<string>Ethernet</string>
		</dict>
	</array>
	<key>__VERSION__</key>
	<integer>20191120</integer>
</dict>
</plist>

And ifconfig:
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> mtu 16384
	options=1203<RXCSUM,TXCSUM,TXSTATUS,SW_TIMESTAMP>
	inet 127.0.0.1 netmask 0xff000000 
	inet6 ::1 prefixlen 128 
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1 
	nd6 options=201<PERFORMNUD,DAD>
gif0: flags=8010<POINTOPOINT,MULTICAST> mtu 1280
stf0: flags=0<> mtu 1280
XHC0: flags=0<> mtu 0
EHC26: flags=0<> mtu 0
EHC29: flags=0<> mtu 0
XHC20: flags=0<> mtu 0
en0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=400<CHANNEL_IO>
	ether 50:3e:aa:8a:8f:e1 
	inet6 fe80::c40:750a:f4c9:141e%en0 prefixlen 64 secured scopeid 0x8 
	inet 192.168.1.18 netmask 0xffffff00 broadcast 192.168.1.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
utun0: flags=8051<UP,POINTOPOINT,RUNNING,MULTICAST> mtu 1380
	inet6 fe80::f0ed:26cb:42ce:4b11%utun0 prefixlen 64 scopeid 0x9 
	nd6 options=201<PERFORMNUD,DAD>
utun1: flags=8051<UP,POINTOPOINT,RUNNING,MULTICAST> mtu 2000
	inet6 fe80::7a36:6deb:4011:ddf7%utun1 prefixlen 64 scopeid 0xa 
	nd6 options=201<PERFORMNUD,DAD>

chris1111, 2020-04-12:
> I tried also to remove my AtherosE2200Ethernet.kext since I was using an ethernet connection before with the same results. My Hackintosh is very laggy, and its unsuable. I also tried different USB [...]

I have 4 USB Wifi then 3 different brand and I don't have any issue like you

If you think this is not your machine and  the Dongle USB Wifi cause problems of your Mac don't use them 
or change the model or change the brand

marcinbauer85, 2020-04-12:
Thanks for the feedbacks. I think I have an USB problem with this dongle, this is the console report:
https://hastebin.com/dexajurejo.sql