chris1111/Wireless-USB-Adapter-Clover #82: No USB-WIFI Plug in

Nishit-Chauhan, 2020-07-11:
i m using catalina 10.15.3, and i have installed latest V15 installed, but not able to use my USB wifi adapter, 
also i have verified the Inject kext = yes in info.plist and after Driver installed also verified that both kext are placed in CLOVER/kext/other/ let me attach screenshot for you. 
i have notice that in system preference no wifi adapter detected in Networks, 
please help me to fix this, Thanks 

[image]
[image]
[image]
[image]

chris1111, 2020-07-11:
What is your adapter brand?

Nishit-Chauhan, 2020-07-11:
https://www.amazon.in/CityShoppy-900Mbps-Wireless-Adapter-Generation/dp/B07FK33GZG

chris1111, 2020-07-14:
Do you have the MacOS drivers?

chromer030, 2020-07-15:
@Nishit-Chauhan , This adapter is based on RTL8188 and on Linux can be loaded via rtl8188fu module , i don't know whether Realtek implemented support for this chip on macOS kexts or not , because this chip is somehow old.

basser2, 2020-07-16:
my problem is that I have to plugin the USB WiFi before my hackintosh boot,otherwise I can't use it. why and how to fix it?

basser2, 2020-07-16:
> i m using catalina 10.15.3, and i have installed latest V15 installed, but not able to use my USB wifi adapter,
> also i have verified the Inject kext = yes in info.plist and after Driver installed [...]

try to plugin the USB WiFi before your system boot?

Nishit-Chauhan, 2020-07-17:
thanks for the response @chris1111 , @chromer030  and @basser2. but actually i have returned this wifi chip, so i m closing this issue

closed 2020-07-17 by Nishit-Chauhan