chris1111/Wireless-USB-Adapter-Clover #87: Diff from official drivers?

francisceril, 2020-08-06:
Not really an issue but what would be the advantage or difference of this driver from the official TP-Link driver? I have the Archer T3U if it matters.
Delete if not appropriate here.

Thanks.

chris1111, 2020-08-06:
It would surprise me that the official TpLink one is more recent than this one

closed 2020-08-06 by chris1111

francisceril, 2020-08-06:
Indeed their driver is still in beta. Anyway, can you provide just the kexts? I'm using OpenCore for my bootloader. The installer I believe is for Clover setup.
Thank you.

chris1111, 2020-08-06:
> Indeed their driver is still in beta. Anyway, can you provide just the kexts? I'm using OpenCore for my bootloader. The installer I believe is for Clover setup.
> Thank you.

https://github.com/chris1111/Wireless-USB-OC-Big-Sur-Adapter