chris1111/Wireless-USB-Adapter-Clover #90: source code.zip , source code.tgz

juma-sayeh, 2020-10-09:
i've tried to download the source code to the project being a mac developer my self, you are not offering any source codes all i saw was the readme.md file and nothing else .. please share codes or remove source code.zip/tgz files

Jude-A, 2020-10-22:
personnaly I just have the Wifi Icon Color Installer with every release and not the driver. I don't understand, if anyone can help me plz.

chris1111, 2020-10-22:
> i've tried to download the source code to the project being a mac developer my self, you are not offering any source codes all i saw was the readme.md file and nothing else .. please share codes or remove source code.zip/tgz files

Did you find any source code of Wifi Realtek drivers for macOS ? if yes tell me were it is😀

closed 2020-10-22 by chris1111