chris1111/Wireless-USB-Adapter-Clover #93: Installation prbl

Jude-A, 2020-10-22:
Hey, I have a little problem, when I download the zip or tar fil and decompressed them, I just fin the wifi color installer and not the real driver. What's the solution ?

chris1111, 2020-10-22:
Download the release 
https://github.com/chris1111/Wireless-USB-Adapter-Clover#download-latest-release--wireless-usb-adapter-clover-v16pkg-

closed 2020-10-22 by chris1111

Jude-A, 2020-10-22:
it does not work

Jude-A, 2020-10-22:
[image]

chris1111, 2020-10-22:
This is a Joke or what You have 9 link for Release
https://github.com/chris1111/Wireless-USB-Adapter-Clover#download-latest-release--wireless-usb-adapter-clover-v16pkg-
If you select the first link its the release V16

[image]