chris1111/Wireless-USB-Adapter-Clover #97: Can't connect to a network

Zetta420, 2021-01-18:
Hello, I got a little problem.
I'm trying to make an hackintosh with an Acer Aspire v5 571g.
Actually I didn't installed it yet, I need some internet. The problem is: I don't have ethernet connection on m'y laptop, I can't find working wifi drivers for the default laptop wifi card, I tryed a kext for Android phones, but not working.
Si I tryed with a tplink adapter.
The adapter is clearly detected by mac os, but I can't connect to wifi. It look like the adapter is recognized as ethernet.
But I need to pick my wifi and put my password, this is my only way to install mac.
[image]
There are the only infos I could get... (sorry for french on pic, I'm french).
If you have any idea/trick...
This is actually the only thing not working (and probably sound, will see After the install. But battery support, trackpad, keyboard are working, camera should but not tested).

Thanks for your help

chris1111, 2021-01-18:
the adapter is recognized as ethernet on new macOS this is normal so this is not an issue.

closed 2021-01-18 by chris1111

Zetta420, 2021-01-18:
Is there any way to dolce my problem knowing this ?

Zetta420, 2021-01-18:
Sorry for double post, my phone autocorrection.
Any way to solve* not dolce

chris1111, 2021-01-18:
> Is there any way to dolce my problem knowing this ?

what problem

Zetta420, 2021-01-18:
As I said, I have no ethernet port on my laptop, I tryed to find my PCI wifi card drivers, but seems to be not working, and tryed with HoRNDIS to connect my phone, and seems to be not working too...

And I need internet connection to install mac

chris1111, 2021-01-18:
Sorry but your internet issue for installing macOS is not relate to the USB adapter on this project the USB WIFI adapter not working for downloading macOS from recovery HD everybody knonws this so you have to put a Mini PCI WIFI compatible macOS