chris1111/Wireless-USB-Adapter-Clover #100: Incompatible with Adobe XD

wdhwg001, 2021-02-15:
Hi,

I'm using a Netgear A6150 with this driver, but the driver stops working once I started Adobe XD. In detail, the icon hangs with no responses and the network disconnected until I close Adobe XD and replug the USB WiFi.

I don't have much experience with debugging such incompatibilities, but I do notice that Adobe XD uses libusb. I suppose it uses that to communicate with the mobile app to provide on-device previews. I'm personally not quite interested in this feature, as being incompatible with a WiFi driver is much more frustrating.

I did post feedback to Adobe, but I suspect they won't care as this driver is pretty much for Hackintosh only. Therefore, is there any way to workaround or fix this here?

Thanks!

chris1111, 2021-02-15:
Why not trying a macOS Update ?

ymc3, 2021-02-15:
> Why not trying a macOS Update ?

you mean upgrade to big sur?

chris1111, 2021-02-15:
No update 10.15.7

closed 2021-03-21 by chris1111