chris1111/Wireless-USB-Adapter-Clover #101: Mac not recognizing the USB-Wifi

guga330, 2021-03-17:
Hello, I have the following specs:
- Macbookpro 2017 15"
- Running MacOS Catalina 10.15.7

My issue is that after installing the driver, it reaches a point where it starts to glitch and turn on and off the wifi until it doesn't recognize the USB-Wifi card (Archer T4U). What fixes the issue is restarting the Mac with the USB plugged in, as soon as it logs in it recognizes the Wifi card and allows me to connect the a Wifi. Usually when the Mac is on and running I have no issues, but when it goes to sleep and I wake it back up, the driver doesn't recognize the USB-Wifi card.
- I have tried different ports, all lead to the same point
- I have done a NVRAM reset
Maybe I didn't follow the steps correctly, but I have no idea what is causing this?
If you could help me out, it would be much appreciated.
Thank you
[image]

chris1111, 2021-03-17:
no native wifi on your Mac?

guga330, 2021-03-17:
> no native wifi on your Mac?

Yes I do have a native Wifi. Except the wifi network I'm trying to reach need the Wifi-USB adapter (archer T4U)

closed 2021-04-22 by chris1111