chris1111/Wireless-USB-Adapter-Clover #112: Installation seems seemingly successful but App not starting

fritzlaszlo, 2022-12-12:
Hi,

- I installed the app on a hackintosh running macOS High Sierra.
- SIP disabled
- gatekeeper disabled. 
- Clover was configured to allow injekt kexts. 
- I find the kext files after an seemingly successful installation and a reboot in my EFI Partition in the "other"-Folder. 
- I have an TP-Link Archer T3U Plus plugged in, with different USB-Ports tested.
- An 802.11ac NIC interface shows up both in my hardware report / system information and my network preferences. - Yet no app is showing up in the menu bar on the top right, so no way to configure an wifi. 
- I have restarted the computer multiple times. 
- Also, this hackintosh doesn't have any "native wifi". The T3U plus was supposed to be its first wifi.

Any help would be very much appreciated. I dont have much ideas from here on apart from re-installing and keep trying different USB Ports. The T3U Plus is brand new.
Thanks so much!

[image]

fritzlaszlo, 2022-12-12:
So, I investigated a little further and found the status bar app in /library/application support/wifi. When trying to open it manually, it says the app required 10.14. I have High Sierra 10.13.6, which is supported according to the list. I opened the app contents and tried manipulating the info.plist where the string says: "LSMinimumSystemVersion" from 10.14 to 10.13, but tit didn't help. Also, after that trying to run the executables in the MacOS subfolder didn't help. I will now try installing older versions of the app/driver and see if it helps.

fritzlaszlo, 2022-12-12:
As a test, I unistalled V16 via the option in the .pkg. then installed v15. Again kext files in the efi folder. This time also the app logo in die status bar, but it apprently didnt find the T3UPlus since it only kept saying "No USB-Wifi Plug in." I deinstalled v15 again via the pkg und tried v14. Same issue :/

fritzlaszlo, 2022-12-13:
@chris1111 Any chance exporting another version of this software that doesn't have this minimum OS Software? Maybe it was just a mistake, since it says 10.13 should be supported in the readme. Thanks anyways for your fantastic work!

chris1111, 2022-12-13:
Thank you for investigate! You are right I have fix the issue
[Download V17](https://github.com/chris1111/Wireless-USB-Adapter-Clover/releases/tag/V17)

Use the V17 then perform an Uninstall then reboot, after reboot Install V17 
Please repport 

[image]

fritzlaszlo, 2022-12-14:
Thanks a lot @chris1111 for the quick fix!
The App no works and starts on bootup. However my T3U Plus still didnt get recognized on multiple USB Ports.
I then tried the "official" TP-Link driver for mac and with their driver/app-combo it works.
[image]

closed 2022-12-14 by chris1111